os-fpga/Raptor #1481: Build error: make shows error

anthonyle-ql, 2024-02-01:
After I installed the '/.github/workflows/install_ubuntu_dependencies_build.sh', the 'make' still shows error. Please see log file.
[build.zip](https://github.com/os-fpga/Raptor/files/14119297/build.zip)

alaindargelas, 2024-02-01:
@ravikiranchollangi , please check

ravikiranchollangi, 2024-02-01:
@anthonyle-ql we need your system information, which version of linux are you on? Are you sure the dependencies were installed successfully ?

@NadeemYaseen there's a java error. Does any of the dependencies bring in the java install ?
CMake Error at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Java (missing: Java_JAVA_EXECUTABLE Runtime) (Required is at
  least version "11")
Call Stack (most recent call first):
  /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.22/Modules/FindJava.cmake:324 (find_package_handle_standard_args)
  third_party/surelog/CMakeLists.txt:280 (find_package)

anthonyle-ql, 2024-02-01:
Hello,

System information:
LSB Version:	core-11.1.0ubuntu4-noarch:security-11.1.0ubuntu4-noarch
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.3 LTS
Release:	22.04
[requirements.zip](https://github.com/os-fpga/Raptor/files/14127347/requirements.zip)

Codename:	jammy

Please see the attached log from installing required packages.

alaindargelas, 2024-02-01:
Yes, Surelog does need java. 
Let's amend the dependencies.

ravikiranchollangi, 2024-02-01:
> Hello,
>
>
>
> System information:
>
> LSB Version:	core-11.1.0ubuntu4-noarch:security-11.1.0ubuntu4-noarch
>
> Distributor ID:	Ubuntu
>
> Description:	Ubuntu 22.04.3 LTS
>
> Release:	22.04 [...]

@anthonyle-ql please install the following and let us know. We will update the dependencies accordingly. 

sudo apt install default-jre

anthonyle-ql, 2024-02-01:
execute: sudo apt install default-jre
result: OK - no issue

Build still generates error. Starting from line 1013, the compilation shows errors. Example of error:
/home/qlblue/anthony/Raptor/Raptor_Tools/parser_plugins/synlig/build/third_party/surelog/third_party/UHDM/generated/uhdm/virtual_interface_var.h:60:79: error: ‘uhdmvirtual_interface_var’ is not a member of ‘UHDM::UHDM_OBJECT_TYPE’
   60 |   virtual  UHDM_OBJECT_TYPE UhdmType() const final { return UHDM_OBJECT_TYPE::uhdmvirtual_interface_var; }

[build.zip](https://github.com/os-fpga/Raptor/files/14128449/build.zip)

alaindargelas, 2024-02-01:
Please follow all the steps here:
https://github.com/chipsalliance/Surelog/blob/master/INSTALL.md

ravikiranchollangi, 2024-02-01:
@anthonyle-ql can you check if your system doesn't have multiple versions of java installed ?

`update-java-alternatives --list`

If yes, can you remove the ones older ones ?
https://www.baeldung.com/linux/java-choose-default-version

anthonyle-ql, 2024-02-01:
Execute: update-java-alternatives --list
Result: only single version of JAVA
qlblue@qlblue-4:~/anthony/Raptor$ update-java-alternatives --list
java-1.11.0-openjdk-amd64      1111       /usr/lib/jvm/java-1.11.0-openjdk-amd64

Follow instruction to install dependencies: sudo apt-get install build-essential cmake git pkg-config tclsh swig uuid-dev libgoogle-perftools-dev python3 python3-orderedmultidict python3-psutil python3-dev default-jre lcov zlib1g-dev
Result: no issue - cmake version is 3.22.1
Reading package lists...
Building dependency tree...
Reading state information...
build-essential is already the newest version (12.9ubuntu3).
default-jre is already the newest version (2:1.11-72build2).
libgoogle-perftools-dev is already the newest version (2.9.1-0ubuntu3).
pkg-config is already the newest version (0.29.2-1ubuntu3).
python3-psutil is already the newest version (5.9.0-1build1).
tcl is already the newest version (8.6.11+1build2).
uuid-dev is already the newest version (2.37.2-4ubuntu3).
lcov is already the newest version (1.15-1).
python3-orderedmultidict is already the newest version (1.0-5).
swig is already the newest version (4.0.2-1ubuntu1).
cmake is already the newest version (3.22.1-1ubuntu1.22.04.1).
git is already the newest version (1:2.34.1-1ubuntu1.10).
python3 is already the newest version (3.10.6-1~22.04).
python3-dev is already the newest version (3.10.6-1~22.04).
zlib1g-dev is already the newest version (1:1.2.11.dfsg-2ubuntu9.2).
0 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.
cmake version 3.22.1
CMake suite maintained and supported by Kitware (kitware.com/cmake).

Execute: make for Raptor tool
Result: same errors as before
[build1.zip](https://github.com/os-fpga/Raptor/files/14129339/build1.zip)

alaindargelas, 2024-02-01:
Restart from scratch now the system has all the packages:

rm -rf Raptor
git clone .....

anthonyle-ql, 2024-02-01:
remove directory, clone and build shows no error. 'build' folder is created and is able to launch raptor GUI. Will proceed with testing. This item can be closed.

closed 2024-02-01 by anthonyle-ql