os-fpga/Raptor #1527: Timing

saaramahmoudi, 2024-02-20:
Hi,

I am trying to use the Raptor tool and am interested in turning on the "timing_flow analytic" option. According to the document, it is supposed to do an analytical placement (as an initial placement) followed by a simulated annealing process (as a refinement step).

Whenever I add this option to the .tcl file, raptor errors in the VPR stage, telling me there is no command line option called "enable_cascade_placer". Then, I examined the code in the original PR called "https://github.com/os-fpga/Backend/pull/338" and saw this option was added to multiple files:

include/read_verilog/OpenFPGA/vpr/src/base/SetupVPR.cpp
include/read_verilog/OpenFPGA/vpr/src/base/read_options.cpp
include/read_verilog/OpenFPGA/vpr/src/base/read_options.h\
include/read_verilog/OpenFPGA/vpr/src/base/vpr_types.h
However, none of the above files exist in the master branch, which might be causing the issue.
Can you please help me to figure out how to use this option? I really appreciate your time and response in advance.

alaindargelas, 2024-02-28:
@serge-dsa is going to re-enable this code in a couple of weeks, or maybe sooner.

serge-dsa, 2024-03-01:
@saaramahmoudi @alaindargelas 
The patch to analytic placer is now restored. But the "enable_cascade_placer" VPR option seems to be a different patch. I'll check.

serge-dsa, 2024-03-05:
The "enable_cascade_placer" VPR option is now restored.

closed 2024-03-05 by serge-dsa

alaindargelas, 2024-03-08:
@bbambhania1 FYI