os-fpga/Raptor #1529: Can not find qt6 package of centos 7

ducminhnguyen123, 2024-02-22:
I haven't found any qt6 packages on centos 7. I followed the guideline in this link: [https://unix.stackexchange.com/questions/766218/installing-qt-6-on-centos-7](url) but itcan not work and I encourtered this error below:

CMake Error at FOEDAG_rs/FOEDAG/cmake/cmake_qt.txt:22 (find_package):
  By not providing "FindQt6.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Qt6", but
  CMake did not find one.

  Could not find a package configuration file provided by "Qt6" with any of
  the following names:

    Qt6Config.cmake
    qt6-config.cmake

  Add the installation prefix of "Qt6" to CMAKE_PREFIX_PATH or set "Qt6_DIR"
  to a directory containing one of the above files.  If "Qt6" provides a
  separate development package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  FOEDAG_rs/FOEDAG/CMakeLists.txt:76 (include)

-- Configuring incomplete, errors occurred!

This is the CMake output log: 
[cmake_qt.txt](https://github.com/os-fpga/Raptor/files/14368369/cmake_qt.txt)
I have used yum search qt to find qt6 and I don't see any qt6 packages.

pls, give me the instructions to install qt6 on centos 7

NadeemYaseen, 2024-02-22:
All the needed packages including Qt6 can be installed via [install_centos_dependencies_build.sh](https://github.com/os-fpga/Raptor/blob/main/.github/workflows/install_centos_dependencies_build.sh)
We also offer CentOS 7 Docker image with all the packages installed. You can pulled it via
```
docker pull ghcr.io/os-fpga/raptor:centos7latest
```

ducminhnguyen123, 2024-02-22:
thanks a lot

ducminhnguyen123, 2024-02-22:
Sorry for this inconvenience but when I follow [install_centos_dependencies_build.sh](https://github.com/os-fpga/Raptor/blob/main/.github/workflows/install_centos_dependencies_build.sh), I have encoutered a new issue related to QT6.
Using Monaco Editor
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE  
-- Check if linker can resolve circular dependencies - FALSE
-- Performing Test HAVE_STDATOMIC
-- Performing Test HAVE_STDATOMIC - Success
-- Found WrapAtomic: TRUE  
-- Could NOT find Qt6CoreTools (missing: Qt6CoreTools_DIR)
CMake Warning at /usr/local/Qt6.2.4/lib/cmake/Qt6/Qt6Config.cmake:209 (find_package):
  Found package configuration file:

    /usr/local/Qt6.2.4/lib/cmake/Qt6Core/Qt6CoreConfig.cmake

  but it set Qt6Core_FOUND to FALSE so package "Qt6Core" is considered to be
  NOT FOUND.  Reason given by package:

  Target "Qt6::Core" was not found.

Call Stack (most recent call first):
  FOEDAG_rs/FOEDAG/cmake/cmake_qt.txt:22 (find_package)
  FOEDAG_rs/FOEDAG/CMakeLists.txt:76 (include)

-- Found OpenGL: /usr/lib64/libOpenGL.so   
-- Found WrapOpenGL: TRUE  
-- Found XKB: /usr/lib64/libxkbcommon.so (found suitable version "0.7.1", minimum required is "0.5.0") 
-- Could NOT find Qt6GuiTools (missing: Qt6GuiTools_DIR)
CMake Warning at /usr/local/Qt6.2.4/lib/cmake/Qt6/Qt6Config.cmake:209 (find_package):
  Found package configuration file:

    /usr/local/Qt6.2.4/lib/cmake/Qt6Gui/Qt6GuiConfig.cmake

  but it set Qt6Gui_FOUND to FALSE so package "Qt6Gui" is considered to be
  NOT FOUND.  Reason given by package:

  Target "Qt6::Gui" was not found.

Call Stack (most recent call first):
  FOEDAG_rs/FOEDAG/cmake/cmake_qt.txt:22 (find_package)
  FOEDAG_rs/FOEDAG/CMakeLists.txt:76 (include)

-- Could NOT find Qt6WidgetsTools (missing: Qt6WidgetsTools_DIR)
CMake Warning at /usr/local/Qt6.2.4/lib/cmake/Qt6/Qt6Config.cmake:209 (find_package):
  Found package configuration file:

    /usr/local/Qt6.2.4/lib/cmake/Qt6Widgets/Qt6WidgetsConfig.cmake

  but it set Qt6Widgets_FOUND to FALSE so package "Qt6Widgets" is considered
  to be NOT FOUND.  Reason given by package:

  Target "Qt6::Widgets" was not found.

Call Stack (most recent call first):
  FOEDAG_rs/FOEDAG/cmake/cmake_qt.txt:22 (find_package)
  FOEDAG_rs/FOEDAG/CMakeLists.txt:76 (include)

-- Could NOT find Qt6CoreTools (missing: Qt6CoreTools_DIR)
CMake Warning at /usr/local/share/cmake-3.16/Modules/CMakeFindDependencyMacro.cmake:47 (find_package):
  Found package configuration file:

    /usr/local/Qt6.2.4/lib/cmake/Qt6Core/Qt6CoreConfig.cmake

  but it set Qt6Core_FOUND to FALSE so package "Qt6Core" is considered to be
  NOT FOUND.  Reason given by package:

  Target "Qt6::Core" was not found.

Call Stack (most recent call first):
  /usr/local/Qt6.2.4/lib/cmake/Qt6/QtPublicDependencyHelpers.cmake:14 (find_dependency)
  /usr/local/Qt6.2.4/lib/cmake/Qt6Xml/Qt6XmlDependencies.cmake:96 (_qt_internal_find_dependencies)
  /usr/local/Qt6.2.4/lib/cmake/Qt6Xml/Qt6XmlConfig.cmake:40 (include)
  /usr/local/Qt6.2.4/lib/cmake/Qt6/Qt6Config.cmake:209 (find_package)
  FOEDAG_rs/FOEDAG/cmake/cmake_qt.txt:22 (find_package)
  FOEDAG_rs/FOEDAG/CMakeLists.txt:76 (include)

CMake Warning at /usr/local/Qt6.2.4/lib/cmake/Qt6/Qt6Config.cmake:209 (find_package):
  Found package configuration file:

    /usr/local/Qt6.2.4/lib/cmake/Qt6Xml/Qt6XmlConfig.cmake

  but it set Qt6Xml_FOUND to FALSE so package "Qt6Xml" is considered to be
  NOT FOUND.  Reason given by package:

  Qt6Xml could not be found because dependency Qt6Core could not be found.

Call Stack (most recent call first):
  FOEDAG_rs/FOEDAG/cmake/cmake_qt.txt:22 (find_package)
  FOEDAG_rs/FOEDAG/CMakeLists.txt:76 (include)

-- Could NOT find Qt6CoreTools (missing: Qt6CoreTools_DIR)
CMake Warning at /usr/local/share/cmake-3.16/Modules/CMakeFindDependencyMacro.cmake:47 (find_package):
  Found package configuration file:

    /usr/local/Qt6.2.4/lib/cmake/Qt6Core/Qt6CoreConfig.cmake

  but it set Qt6Core_FOUND to FALSE so package "Qt6Core" is considered to be
  NOT FOUND.  Reason given by package:

  Target "Qt6::Core" was not found.

I have installed all packages except the cmake, cmake is just cmake 16. I wonder it is the cause of issue

closed 2024-02-22 by ducminhnguyen123