os-fpga/Raptor #1530: issue with qt6

ducminhnguyen123, 2024-02-22:
Sorry for this inconvenience but when I follow [install_centos_dependencies_build.sh](https://github.com/os-fpga/Raptor/blob/main/.github/workflows/install_centos_dependencies_build.sh), I have encoutered a new issue related to QT6.
Using Monaco Editor
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE
-- Check if linker can resolve circular dependencies - FALSE
-- Performing Test HAVE_STDATOMIC
-- Performing Test HAVE_STDATOMIC - Success
-- Found WrapAtomic: TRUE
-- Could NOT find Qt6CoreTools (missing: Qt6CoreTools_DIR)
CMake Warning at /usr/local/Qt6.2.4/lib/cmake/Qt6/Qt6Config.cmake:209 (find_package):
Found package configuration file:

/usr/local/Qt6.2.4/lib/cmake/Qt6Core/Qt6CoreConfig.cmake
but it set Qt6Core_FOUND to FALSE so package "Qt6Core" is considered to be
NOT FOUND. Reason given by package:

Target "Qt6::Core" was not found.

Call Stack (most recent call first):
FOEDAG_rs/FOEDAG/cmake/cmake_qt.txt:22 (find_package)
FOEDAG_rs/FOEDAG/CMakeLists.txt:76 (include)

-- Found OpenGL: /usr/lib64/libOpenGL.so
-- Found WrapOpenGL: TRUE
-- Found XKB: /usr/lib64/libxkbcommon.so (found suitable version "0.7.1", minimum required is "0.5.0")
-- Could NOT find Qt6GuiTools (missing: Qt6GuiTools_DIR)
CMake Warning at /usr/local/Qt6.2.4/lib/cmake/Qt6/Qt6Config.cmake:209 (find_package):
Found package configuration file:

/usr/local/Qt6.2.4/lib/cmake/Qt6Gui/Qt6GuiConfig.cmake
but it set Qt6Gui_FOUND to FALSE so package "Qt6Gui" is considered to be
NOT FOUND. Reason given by package:

Target "Qt6::Gui" was not found.

Call Stack (most recent call first):
FOEDAG_rs/FOEDAG/cmake/cmake_qt.txt:22 (find_package)
FOEDAG_rs/FOEDAG/CMakeLists.txt:76 (include)

-- Could NOT find Qt6WidgetsTools (missing: Qt6WidgetsTools_DIR)
CMake Warning at /usr/local/Qt6.2.4/lib/cmake/Qt6/Qt6Config.cmake:209 (find_package):
Found package configuration file:

/usr/local/Qt6.2.4/lib/cmake/Qt6Widgets/Qt6WidgetsConfig.cmake
but it set Qt6Widgets_FOUND to FALSE so package "Qt6Widgets" is considered
to be NOT FOUND. Reason given by package:

Target "Qt6::Widgets" was not found.

Call Stack (most recent call first):
FOEDAG_rs/FOEDAG/cmake/cmake_qt.txt:22 (find_package)
FOEDAG_rs/FOEDAG/CMakeLists.txt:76 (include)

-- Could NOT find Qt6CoreTools (missing: Qt6CoreTools_DIR)
CMake Warning at /usr/local/share/cmake-3.16/Modules/CMakeFindDependencyMacro.cmake:47 (find_package):
Found package configuration file:

/usr/local/Qt6.2.4/lib/cmake/Qt6Core/Qt6CoreConfig.cmake
but it set Qt6Core_FOUND to FALSE so package "Qt6Core" is considered to be
NOT FOUND. Reason given by package:

Target "Qt6::Core" was not found.

Call Stack (most recent call first):
/usr/local/Qt6.2.4/lib/cmake/Qt6/QtPublicDependencyHelpers.cmake:14 (find_dependency)
/usr/local/Qt6.2.4/lib/cmake/Qt6Xml/Qt6XmlDependencies.cmake:96 (_qt_internal_find_dependencies)
/usr/local/Qt6.2.4/lib/cmake/Qt6Xml/Qt6XmlConfig.cmake:40 (include)
/usr/local/Qt6.2.4/lib/cmake/Qt6/Qt6Config.cmake:209 (find_package)
FOEDAG_rs/FOEDAG/cmake/cmake_qt.txt:22 (find_package)
FOEDAG_rs/FOEDAG/CMakeLists.txt:76 (include)

CMake Warning at /usr/local/Qt6.2.4/lib/cmake/Qt6/Qt6Config.cmake:209 (find_package):
Found package configuration file:

/usr/local/Qt6.2.4/lib/cmake/Qt6Xml/Qt6XmlConfig.cmake
but it set Qt6Xml_FOUND to FALSE so package "Qt6Xml" is considered to be
NOT FOUND. Reason given by package:

Qt6Xml could not be found because dependency Qt6Core could not be found.

Call Stack (most recent call first):
FOEDAG_rs/FOEDAG/cmake/cmake_qt.txt:22 (find_package)
FOEDAG_rs/FOEDAG/CMakeLists.txt:76 (include)

-- Could NOT find Qt6CoreTools (missing: Qt6CoreTools_DIR)
CMake Warning at /usr/local/share/cmake-3.16/Modules/CMakeFindDependencyMacro.cmake:47 (find_package):
Found package configuration file:

/usr/local/Qt6.2.4/lib/cmake/Qt6Core/Qt6CoreConfig.cmake
but it set Qt6Core_FOUND to FALSE so package "Qt6Core" is considered to be
NOT FOUND. Reason given by package:

Target "Qt6::Core" was not found.

I have installed all packages except the cmake, cmake is just cmake 16. I wonder it is the cause of issue @NadeemYaseen and how can I fix this issue

NadeemYaseen, 2024-02-22:
It is recommended to have cmake version greater than 20 as down the path you will face the issue with cmake version 16.

Regarding the Qt6 issue, I believe now you have multiple installations of Qt so try exporting the below variables 

```
export QMAKE_CC='<path to GCC 11 compiler>'
export QMAKE_CXX='<path to GCC G++ 11 compiler>'
export PATH="/usr/local/Qt6.2.4/bin:/usr/lib/ccache:$PATH"  # assuming you have Qt 6.2.4 in /usr/local/
export Qt6_DIR=/usr/local/Qt6.2.4   # assuming you have Qt 6.2.4 in /usr/local/
```

Also, make sure Qt 6 has all dependent libs completely installed, One way is to check the dynamic linking via `ldd` command like below

```
ldd -r /usr/local/Qt6.2.4/lib/Qt6Core.so.6
```

ducminhnguyen123, 2024-02-23:
thanks a lot, the flow is running smoothly when I make. But when I run the the instruction:
`make test/rs_gui`
an error appears:
LINUX MODE
-- Configuring done (4.7s)
-- Generating done (2.0s)
-- Build files have been written to:/home/ductiger1/Documents/Raptor/build
./build/bin/raptor --script tests/Testcases/aes_decrypt_fpga/aes_decrypt.tcl
./build/bin/raptor: line 12:  3253 Aborted                 (core dumped) $RAPTOR_PATH/bin/raptor.exe "$@" 2> /dev/null
make: *** [Makefile:143: test/rs_gui] Error 134
Can you show me the instructions to fix this issue?

ducminhnguyen123, 2024-02-23:
thanks a lot

closed 2024-02-23 by ducminhnguyen123