os-fpga/Raptor #1534: Run test fail

ducminhnguyen123, 2024-02-23:
the flow is running smoothly when I make. But when I run the the instruction:
make test/rs_gui
an error appears:
LINUX MODE
-- Configuring done (4.7s)
-- Generating done (2.0s)
-- Build files have been written to:/home/ducminhnguyen123/Documents/Raptor/build
./build/bin/raptor --script tests/Testcases/aes_decrypt_fpga/aes_decrypt.tcl
./build/bin/raptor: line 12: 3253 Aborted (core dumped) $ RAPTOR_PATH/bin/raptor.exe "$@" 2> /dev/null
make: *** [Makefile:143: test/rs_gui] Error 134
Can you show me the instructions to fix this issue?

NadeemYaseen, 2024-02-23:
Try to run the below test case:
```
./build/bin/raptor --batch --script tests/tcl_examples/and2_verilog/run_raptor.tcl --device MPW1 
```
This will run the batch mode of Raptor.

For the GUI of Raptor, I need information about how you are getting the Display of your CentOS 7 like it is a remote server and you accessing it via some NoMachine or it is your VM.

ducminhnguyen123, 2024-02-23:
after using your given test case, the Raptor run well. I am using my main computer (not using any remote servers or VM) to execute Raptor (CentOS 7 is my main os of my computer).

ducminhnguyen123, 2024-02-26:
this is the output of .build/bin/raptor:
#!/bin/bash
# Raptor environment setup script - 64-bit Linux
# Copyright(c) 2022 Rapid Silicon
# All licenses available in $RAPTOR_PATH/share/raptor/licenses
# Version 1.0

RETURN_PATH=`pwd`
SCRIPT_PATH=`dirname $BASH_SOURCE`
RAPTOR_PATH=`( cd "$SCRIPT_PATH" && pwd )`

[ -f $RAPTOR_PATH/../.raptorenv_lin64.sh ] && source $RAPTOR_PATH/../.raptorenv_lin64.sh
$RAPTOR_PATH/bin/raptor.exe "$@" 2>/dev/null

NadeemYaseen, 2024-02-26:
In file `./build/bin/raptor`  change the last line from `$RAPTOR_PATH/bin/raptor.exe "$@" 2>/dev/null` to `$RAPTOR_PATH/bin/raptor.exe "$@"`
After this running `./build/bin/raptor` should print the missing libraries. Please share the output.

ducminhnguyen123, 2024-02-26:
this is the output(I copy the output from terminal and paste to this file):
[output.txt](https://github.com/os-fpga/Raptor/files/14402967/output.txt)

NadeemYaseen, 2024-02-26:
From output.txt, I can see below:
```
src/tcmalloc.cc:284] Attempt to free invalid pointer 0x3a30960
```
This error is reported by many GUI based application specially on CentOS 7.

You can compile the tcmalloc version `4.5.10` or later. 
Or
I am also attaching this library (compiled). You can unzip it and place it in `./build/lib/openssl` directory. Then try to run the Raptor.
[tcmollacForCentos.tar.gz](https://github.com/os-fpga/Raptor/files/14404120/tcmollacForCentos.tar.gz)

ducminhnguyen123, 2024-02-27:
I used your [tcmollacForCentos.tar.gz](https://github.com/os-fpga/Raptor/files/14404120/tcmollacForCentos.tar.gz) and do as the guideline but it didn't work.

NadeemYaseen, 2024-02-27:
remove the tcmollacForCentOS level. Place the libs directly under openssl directory
e.g.

```
build/lib/openssl/<all the libs>
```

ducminhnguyen123, 2024-02-28:
Because of your instructions, the current issue disappeared. But, a new issue has appeared. I put the output in this file
[new_output.txt](https://github.com/os-fpga/Raptor/files/14427572/new_output.txt)
I don't know the reason why I have encoutered this issue. Or is there an output log (which I don't know)? Can you give me the instructions for this issue, pls? @NadeemYaseen

alaindargelas, 2024-02-28:
@ducminhnguyen123, the only targets you should use are:
**make 
make debug
make install**

You used:
make test/rs_gui
....
make: *** [test/rs_gui] Error 127

Please use only the 3 targets above.

Once the tool is built, simply run the tests individually and make sure you target your device.
When you are at the point where you want to make a contribution (Pull Request), we will review the pull request, there are a few tests that are running part of the open-source CI, and we have 1000s of tests that are running privately so we can test your change.

alaindargelas, 2024-02-28:
@ducminhnguyen123 I see where the problem comes from, the BUILD.md instructions needed to be updated.
I just did that. Please pull.
You can now test the build:
make test/raptor_gui
or
make test/ratpor_batch

NadeemYaseen, 2024-02-28:
Error 127 means you are again missing some OS library.
If you have latest code then export below variable to turn on OS dependencies verbosity
```
export RAPTOR_EXE_DEBUG=1
```
Also note that doing  

```
make 
```
is enough to build Raptor. To invoke GUI, just do

```
./build/bin/raptor
```
And to invoke in batch mode

```
./build/bin/raptor --batch
```

ducminhnguyen123, 2024-02-29:
yes you can close it, the reason I have said from the issue: #1546

ducminhnguyen123, 2024-02-29:
make test/raptor_gui: it doesn't work.
make test/raptor_batch: it doesn't work, either.
./build/bin/raptor --batch --script tests/tcl_examples/and2_verilog/run_raptor.tcl --device MPW1: it works well but after I use ./build/bin/raptor and ./build/bin/raptor --batch, I re-run the command ./build/bin/raptor --batch --script tests/tcl_examples/and2_verilog/run_raptor.tcl --device MPW1 and it hasn't worked anymore.
Can you guide me how to can run again the command ./build/bin/raptor --batch --script tests/tcl_examples/and2_verilog/run_raptor.tcl --device MPW1 without re-clone Raptor from scratch.

If the command can run well, I think we can close this issue because at the moment, I want my design can run with Raptor tool and the command ./build/bin/raptor --batch --script tests/tcl_examples/and2_verilog/run_raptor.tcl --device MPW1 can do it

alaindargelas, 2024-02-29:
OK, but it looks like you still have QT Library installation issues on your machine. Closing.

closed 2024-02-29 by alaindargelas