os-fpga/Raptor #1542: Issue with setting device size

ducminhnguyen123, 2024-02-26:
Hi, I have an issue when setting device size: 
[image]
This my command run: 
[image]
I need to do this because when I run the commnad below, an issue has appeared:
[image]
and the device_size default here is 16x16. This value is in my packing.rpt

alaindargelas, 2024-02-27:
@ducminhnguyen123 the layout 50x50 does not exist in the vpr.xml for that device:
https://github.com/os-fpga/Raptor/blob/4f32d13a493162fa1b6f37f3633716e9f2c8446c/etc/devices/mpw1/TSMC22nm_vpr.xml#L2178

You need to create a new device in the device.xml file, make a hard copy of all the files under 
https://github.com/neulink-semi/Raptor/tree/main/etc/devices/mpw1
in a different directory in your fork, then create layouts and do other kind of edits and reference the new device directory path in the device.xml file.
The mpw1 device is given as an example only. It is not the actual device manufactured by Rapid Silicon.
There are many other architecture files in the repo https://github.com/lnis-uofu/OpenFPGA

closed 2024-02-27 by alaindargelas

ducminhnguyen123, 2024-02-27:
tks a lot, it is running well

ducminhnguyen123, 2024-02-28:
Can you give me the instructions to use some options like using options in OpenFPGA ( I see some options in this link: [https://docs.verilogtorouting.org/en/latest/vpr/command_line_usage/](url) and base on this, I can use options as I want in the file which have .openfpga extension 
[image]

alaindargelas, 2024-02-28:
Please read the user guide:
https://github.com/os-fpga/Raptor/blob/main/docs/Raptor_User_Guide.pdf

Also, please open a new issue everytime you have a problem. Please do not keep adding new issues to a closed issue. 
VPR options setting are discussed in the user guide.

The OpenFPGA script (Used only for bitstream generation) is typically not changeable in Raptor, although there are 2 expert-level commands to change it (they are not in the manual), 
https://github.com/os-fpga/FOEDAG/blob/4128cc8299219e0b18d877e02bdffc131aea38cd/src/Compiler/CompilerOpenFPGA.cpp#L282
https://github.com/os-fpga/FOEDAG/blob/4128cc8299219e0b18d877e02bdffc131aea38cd/src/Compiler/CompilerOpenFPGA.cpp#L321

but I don't think that it is what you are after. I think you want to change VPR options which is described in the manual.

ducminhnguyen123, 2024-02-28:
Sorry for my mistake and I will create a new issue