os-fpga/Raptor #1546: Can not work with the Raptor Design Suite

ducminhnguyen123, 2024-02-28:
Hello, I have encountered an issue with doing as the guideline in **1.2.2.1 Working with the Raptor Design Suite** of Raptor User Guide Version 0.6.8.pdf. 
[image].
I can't open GUI or enter the tcl prompt when running raptor --batch (.build/bin/raptor --batch).
But I can run this command on command prompt:
`./build/bin/raptor --batch --script tests/tcl_examples/and2_verilog/run_raptor.tcl --device MPW1 `

alaindargelas, 2024-02-28:
It looks like you either have to re-clone Raptor from scratch and rebuild (Which I recommend) or update your current clone/fork.
The user guide does not have a version in the file name. From your explanation above, it looks like you have an obsolete version.
This is the current file:
https://github.com/os-fpga/Raptor/blob/main/docs/Raptor_User_Guide.pdf

If you created a fork, sync your fork and re-clone your fork.
If you cloned directly, simply re-clone from scratch and apply the library changes @NadeemYaseen gave you.

In Batch mode you should see:
./build/bin/raptor --batch
% puts Hello
Hello
% exit

ducminhnguyen123, 2024-02-28:
I have re-cloned Raptor as you said but I encouterd the same issue. I don't know where the error is or the things you want to see to find out the problem. Also, after re-cloning new Raptor, I run `./build/bin/raptor --batch` then `puts Hello`, the output is `bash: puts: command not found...`.

alaindargelas, 2024-02-28:
@nadeemyaseen-rs it looks like the TCL interpreter does not start nor does it produces errors.

@ducminhnguyen123 ,
How about with the command
make test/raptor_gui

Do you see the gui working?

And 

make test/raptor_batch

Does it run in batch mode, do you see any output?

alaindargelas, 2024-02-28:
@ducminhnguyen123 from this issue: https://github.com/os-fpga/Raptor/issues/1547
It looks like you can run Ratpor.
Can we close this issue #1546?

ducminhnguyen123, 2024-02-29:
make test/raptor_gui: I don't see the GUI of Raptor.

ducminhnguyen123, 2024-02-29:
`make test/raptor_gui`: it doesn't work.
`make test/raptor_batch`: it doesn't work, either.
`./build/bin/raptor --batch --script tests/tcl_examples/and2_verilog/run_raptor.tcl --device MPW1`: it works well but after I use `./build/bin/raptor` and `./build/bin/raptor --batch`, I re-run the command `./build/bin/raptor --batch --script tests/tcl_examples/and2_verilog/run_raptor.tcl --device MPW1` and it hasn't worked anymore.
Can you guide me how to can run again the command `./build/bin/raptor --batch --script tests/tcl_examples/and2_verilog/run_raptor.tcl --device MPW1` without re-clone Raptor from scratch. 

If the command can run well, I think we can close this issue because at the moment, I want my design can run with Raptor tool and the command `./build/bin/raptor --batch --script tests/tcl_examples/and2_verilog/run_raptor.tcl --device MPW1` can do it

closed 2024-02-29 by alaindargelas