os-fpga/Raptor #1547: Options to get rid of route_chan_width and use finding "min route channel with" function of OpenFPGA

ducminhnguyen123, 2024-02-28:
Hello, I want to get rid of route_chan_width options to let Raptor find the minimum route channel width". Because inside Raptor, there is OpenFPGA which have 2 functions: one function is assigning a route channel width by a value, execute with the route channel width and the other function is finding the minimum channel width. As implemting OpenFPGA, if I don't set the "min route channel with" option, it will find the minimum channel width (as default).

I must do this because my RTL design need the channel width is more than 200 but I don't know the minimum channel width for my design.

`VPR was run with the following command-line:
/data/raptor/Raptor/build/bin/vpr /data/raptor/Raptor/build/share/raptor/etc/devices/mpw1/TSMC22nm_vpr.xml /data/raptor/run_raptor/counter_verilog/dti_dma/run_1/synth_1_1/synthesis/dti_dma_post_synth.eblif --sdc_file /data/raptor/run_raptor/counter_verilog/dti_dma/run_1/synth_1_1/impl_1_1_1/packing/dti_dma_openfpga.sdc --route_chan_width 200 --suppress_warnings check_rr_node_warnings.log,check_rr_node --clock_modeling ideal --absorb_buffer_luts off --skip_sync_clustering_and_routing_results off --constant_net_method route --post_place_timing_report dti_dma_post_place_timing.rpt --device 50x50 --allow_unrelated_clustering on --gen_post_synthesis_netlist on --post_synth_netlist_unconn_inputs gnd --inner_loop_recompute_divider 1 --max_router_iterations 1500 --timing_report_detail detailed --timing_report_npaths 100 --top dti_dma --net_file /data/raptor/run_raptor/counter_verilog/dti_dma/run_1/synth_1_1/impl_1_1_1/packing/dti_dma_post_synth.net --place_file /data/raptor/run_raptor/counter_verilog/dti_dma/run_1/synth_1_1/impl_1_1_1/placement/dti_dma_post_synth.place --route_file /data/raptor/run_raptor/counter_verilog/dti_dma/run_1/synth_1_1/impl_1_1_1/routing/dti_dma_post_synth.route --route`

alaindargelas, 2024-02-29:
I don't think there is a good reason to add the support for architecture exploration in Raptor itself. That is what VPR is for, so you should run VPR outside Raptor with the following command (Omitting the --route_chan_width 200 option and the --route option):

/data/raptor/Raptor/build/bin/vpr /data/raptor/Raptor/build/share/raptor/etc/devices/mpw1/TSMC22nm_vpr.xml /data/raptor/run_raptor/counter_verilog/dma/run_1/synth_1_1/synthesis/dma_post_synth.eblif --sdc_file /data/raptor/run_raptor/counter_verilog/dma/run_1/synth_1_1/impl_1_1_1/packing/dma_openfpga.sdc  --suppress_warnings check_rr_node_warnings.log,check_rr_node --clock_modeling ideal --absorb_buffer_luts off --skip_sync_clustering_and_routing_results off --constant_net_method route --post_place_timing_report dma_post_place_timing.rpt --device 50x50 --allow_unrelated_clustering on --gen_post_synthesis_netlist on --post_synth_netlist_unconn_inputs gnd --inner_loop_recompute_divider 1 --max_router_iterations 1500 --timing_report_detail detailed --timing_report_npaths 100 --top dma --net_file /data/raptor/run_raptor/counter_verilog/dma/run_1/synth_1_1/impl_1_1_1/packing/dma_post_synth.net --place_file /data/raptor/run_raptor/counter_verilog/dma/run_1/synth_1_1/impl_1_1_1/placement/dma_post_synth.place --route_file /data/raptor/run_raptor/counter_verilog/dma/run_1/synth_1_1/impl_1_1_1/routing/dma_post_synth.route 

Read from the log what VPR uses as a channel_width, then you have 2 choices to specify it for Raptor in the next run, either:
 - edit etc/device.xml and add the channel_width entry along with the number given by VPR:
```
  <device name="MPW1" series="Gemini" family="Internal-Gemini" package="SBG484" pin_count="44" speedgrade="1" core_voltage="0.8V">
        <resource type="io" num="44"/>
        <resource type="lut" num="160"/>
        <resource type="ff" num="320"/>
        <resource type="bram" num="0"/>
        <resource type="dsp" num="0"/>
        <resource type="channel_width" num="PUT_VPR_CHANNEL_WIDTH_HERE"/>
```
then type make for the change to take effect before you run raptor

- or add the following command in the raptor.tcl file: 
```
...
set_channel_width PUT_VPR_CHANNEL_WIDTH_HERE
packing
place
route
```

ducminhnguyen123, 2024-02-29:
thanks a lot. Can I add you just one more question: I wonder why you add option --route_chan_width as the default when executing while OpenFPGA don't use it as default.

alaindargelas, 2024-02-29:
Raptor is not an architecture exploration tool. VPR and OpenFPGA are.
When you are in Raptor, you target real devices (FPGA or eFPGA) that have already gone through the process of the architecture exploration. Channel width, connectivity patterns, layout resource mix, have already been decided and set in stone.
So there is no point to let VPR/OpenFPGA decide of any parameter at Raptor Compile time. All parameters are fixed.