os-fpga/Raptor #1699: Unable to locate Qt6 packages

MarkoMk, 2024-05-06:
- I have been trying to install Raptor on Ubuntu 20.04 and have been following the [install_ubuntu_dependencies_build.sh](https://github.com/os-fpga/Raptor/blob/main/.github/workflows/install_ubuntu_dependencies_build.sh). Whenever it gets to the following packages, I get a message that it can't locate them. The packages are: qt6-webengine-dev, qt6-webengine* & libqt6webenginecore6* (see picture bellow). Would you mind tellig me what I am doing wrong? Thank you in advance.

[image]

alaindargelas, 2024-05-06:
@MarkoMk , please migrate to Ubuntu 22.04 or later.
We don't support 20.04 or 21.04 anymore.

MarkoMk, 2024-05-07:
Thank you for the swift answer. Sorry for taking up your time!

closed 2024-05-07 by alaindargelas