os-fpga/Raptor #1867: Raptor crash case with Pin Planner

w0lek, 2024-07-09:
The problem was found during the experimenting, if Pin Planner pick up newly added port from HDL.

Steps to reproduce:

- open example project and2_gemini
- checkin Tools/Pin Planner, the two new tabs: "IO Ports" and "Package Pins" are shown
- open and2.v file in editor, and add new port input, for example add

`input a_; `
inside the module and2

was

```
module and2 (
  input a,
  input b,
  input clk,
  input reset,
  output reg c = 1'b0
);

```

become

```
module and2 (
  input a,
  **input a_,**
  input b,
  input clk,
  input reset,
  output reg c = 1'b0
);
```
- save and2.v in editor
**- uncheck the Tools/Pin Planner action, the Raptor will be crashed here.**
- when we re-open Raptor again, and activate Tools/Pin Planner, it will pick up the new a_ port symbol.

ravikiranchollangi, 2024-07-09:
@volodymyrkochyn FYI

volodymyrkochyn, 2024-07-11:
I can reproduce this with just open and close Pin Planner steps

volodymyrkochyn, 2024-07-11:
Fix here: https://github.com/os-fpga/FOEDAG/pull/1654

volodymyrkochyn, 2024-07-16:
@w0lek This is done. Please verify the issue.

closed 2024-07-16 by volodymyrkochyn