os-fpga/Raptor #1933: Cmake FOEDAG_RS compile error:   Failed to find Qt component "WebEngineWidgets".

Martoni, 2024-08-20:
Using `make` command I get this cmake error:

```bash
$ make

...

###########################
FOEDAG_RS_RECURSIVE_SUBSYTEMS
###########################
CMAKE_SYSTEM_NAME: Linux
Using Monaco Editor
Enable Interactive Path Analysis Tool
-- Could NOT find Qt6WebEngineWidgets (missing: Qt6WebEngineWidgets_DIR)
CMake Error at FOEDAG_rs/FOEDAG/cmake/cmake_qt.txt:22 (find_package):
  Found package configuration file:

    /usr/lib/x86_64-linux-gnu/cmake/Qt6/Qt6Config.cmake

  but it set Qt6_FOUND to FALSE so package "Qt6" is considered to be NOT
  FOUND.  Reason given by package:

  Failed to find Qt component "WebEngineWidgets".

  Expected Config file at
  "/usr/lib/x86_64-linux-gnu/cmake/Qt6WebEngineWidgets/Qt6WebEngineWidgetsConfig.cmake"
  does NOT exist

Call Stack (most recent call first):
  FOEDAG_rs/FOEDAG/CMakeLists.txt:86 (include)
```

My system is :
```bash
$ cat /etc/lsb-release 
DISTRIB_ID=LinuxMint
DISTRIB_RELEASE=21
DISTRIB_CODENAME=vanessa
DISTRIB_DESCRIPTION="Linux Mint 21 Vanessa"
```

With linux kernel : 

```bash
$ uname -a
Linux 5.15.0-89-generic #99-Ubuntu SMP Mon Oct 30 20:42:41 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
```

alaindargelas, 2024-08-20:
Follow scrupulously: https://github.com/os-fpga/Raptor/blob/main/BUILD.md , including installing the dependencies.
The Only "make" command you should type is in the "Raptor" directory, not the FOEDAG_rs.

Martoni, 2024-08-21:
Ok I didn't see the link to dependencies sorry

closed 2024-08-21 by Martoni